plieningerweb/esp8266-software-uart #1: Adding this to Arduino 8266

plerup, 2015-07-06:
Is it possible to add this to the Arduino 8266 project (https://github.com/esp8266/Arduino) and if so, how? Thanks and BR /Peter Lerup

plieningerweb, 2015-07-07:
Hey Peter,

thank you for the idea. 
I think it is possible, for sure. Unfortunately, right now I do not have the time to port and test it. Maybe later in August this year. Did you try to run it on EspArduino? Let me know, if you get any further with SoftwareSerial and EspArduino! :+1:

H4M3DR, 2015-08-15:
Hi
actually I need this library to work in arduino ide.

surfcoders, 2015-09-23:
Same here.  It would be wonderful if it can work on the Arduino ESP8266 IDE.  I tried adding the c and h files to the same local path of my INO Arduino project, referenced the h file there and it gives me compile errors (first one is "ets_sys.h:149:32: error: unknown type name 'size_t'".  My guess is that there are some additional library headers needed in order to run under the Arduino IDE.

plerup, 2015-09-23:
I have just made an implementation of SoftwareSerial for the Arduino ESP8266. Try it if you like from:
https://github.com/plerup/espsoftwareserial

surfcoders, 2015-09-25:
Thank you so much!!! Works great!!!  Keep you posted of any issues.

closed 2016-03-01 by plieningerweb