plieningerweb/esp8266-software-uart #2: Rx not functioning

mwolfeu, 2015-07-21:
I am on the latest SDK (pfalcon/esp-open-sdk)
All lines with "PIN_PULLDWN_DIS" are commented out since the new SDK removed it.
(apparently there was a question of whether the ESP even had pulldowns)
compiled fine
Tx works fine
Rx fails

I instrumented user_main.c a bit but I don't see anything that is obviously wrong.

user_main.c
```
#include "softuart.h"

#define user_procTaskPrio        0
#define user_procTaskQueueLen    1
os_event_t    user_procTaskQueue[user_procTaskQueueLen];

//create global softuart instances
Softuart softuart;
Softuart softuart2;
//-------------------------------------------------------------------------------------------------
// loop function will be execute by "os" periodically
static void ICACHE_FLASH_ATTR  loop(os_event_t *events)
{
	char buffer_char;

	//print something on uart_0 to see progress
	os_printf(".");

	if (Softuart_Available(&softuart)) {
	  buffer_char = Softuart_Read(&softuart);  
	  if (softuart.buffer.receive_buffer_head == softuart.buffer.receive_buffer_tail)
            os_printf("Buffer empty\n");
          else
            os_printf("Buffer idx: %d %d\n", softuart.buffer.receive_buffer_head, softuart.buffer.receive_buffer_tail);

          os_printf("Softuart 1 Rx: %d\r\n", buffer_char);
          //Softuart_Putchar(&softuart,buffer_char);
	}
	
	//write example output to softuart 1
	Softuart_Putchar(&softuart,'1');

	//some delay until we run this task again
        os_delay_us(100000);

	// run (schedule) this loop task again 
        system_os_post(user_procTaskPrio, 0, 0 );
}

//-------------------------------------------------------------------------------------------------
//Init function
void ICACHE_FLASH_ATTR  user_init()
{
	// Initialize UART0 to use as debug
        uart_div_modify(0, UART_CLK_FREQ / 9600);
        os_delay_us(100000);

	//init software uart
	Softuart_SetPinRx(&softuart,12);	
	Softuart_SetPinTx(&softuart,14);

	//startup
	Softuart_Init(&softuart,9600);

    //Start our loop task
    system_os_task(loop, user_procTaskPrio,user_procTaskQueue, user_procTaskQueueLen);
    system_os_post(user_procTaskPrio, 0, 0 );
}
```
 
This is the output:
Several contiguous dots = my loopback wire not attached
otherwise when it prints, it was attached
I noticed also  each run of the loop seemed to put >1 chars into the Rx buffer.
Is the a timing issue?  New SDK issue?

```
 �tX�BEx��z���I$b[�8��YHMC� ��SOFTUART initialize gpio
SOFTUART bit_time is 104
SOFTUART RX INIT DONE
SOFTUART TX INIT DONE
SOFTUART INIT DONE
mode : sta(xx:xx:xx:xx:xx:xx)
add if0
f 0, ..........................................scandone
.reconnect
f 0, ..........................................scandone
.reconnect
f -240, .Buffer idx: 1 3
Softuart 1 Rx: 0
.Buffer idx: 2 6
Softuart 1 Rx: 0
.Buffer idx: 3 9
Softuart 1 Rx: 0
.Buffer idx: 4 12
Softuart 1 Rx: 0
.Buffer idx: 5 15
Softuart 1 Rx: 0
.Buffer idx: 6 18
Softuart 1 Rx: 0
.Buffer idx: 7 21
Softuart 1 Rx: 0
.Buffer idx: 8 24
Softuart 1 Rx: 0
.Buffer idx: 9 27
Softuart 1 Rx: 0
.Buffer idx: 10 30
Softuart 1 Rx: 0
.Buffer idx: 11 33
Softuart 1 Rx: 0
.Buffer idx: 12 36
Softuart 1 Rx: 0
.Buffer idx: 13 39
Softuart 1 Rx: 0
.Buffer idx: 14 42
Softuart 1 Rx: 0
.Buffer idx: 15 45
Softuart 1 Rx: 0
.Buffer idx: 16 48
Softuart 1 Rx: 0
.Buffer idx: 17 51
Softuart 1 Rx: 0
.Buffer idx: 18 54
Softuart 1 Rx: 0
.Buffer idx: 19 57
Softuart 1 Rx: 0
.Buffer idx: 20 60
Softuart 1 Rx: 0
.Buffer idx: 21 63
Softuart 1 Rx: 0
.Buffer idx: 22 2
Softuart 1 Rx: 0
.Buffer idx: 23 5
Softuart 1 Rx: 0
.Buffer idx: 24 8
Softuart 1 Rx: 0
.Buffer idx: 25 11
Softuart 1 Rx: 0
.Buffer idx: 26 14
Softuart 1 Rx: 0
.Buffer idx: 27 17
Softuart 1 Rx: 0
.Buffer idx: 28 20
Softuart 1 Rx: 0
.Buffer idx: 29 23
Softuart 1 Rx: 0
.Buffer idx: 30 26
Softuart 1 Rx: 0
.Buffer idx: 31 29
Softuart 1 Rx: 0
.Buffer idx: 32 30
Softuart 1 Rx: 0
.Buffer idx: 33 31
Softuart 1 Rx: 0
.Buffer idx: 34 32
Softuart 1 Rx: 0
.Buffer idx: 35 32
Softuart 1 Rx: 0
.Buffer idx: 36 32
Softuart 1 Rx: 0
.Buffer idx: 37 32
Softuart 1 Rx: 0
.Buffer idx: 38 32
Softuart 1 Rx: 0
.Buffer idx: 39 32
Softuart 1 Rx: 0
.Buffer idx: 40 32
Softuart 1 Rx: 0
.Buffer idx: 41 32
Softuart 1 Rx: 0
.Buffer idx: 42 32
Softuart 1 Rx: 0
scandone
.Buffer idx: 43 32
Softuart 1 Rx: 0
reconnect
f 0, .Buffer idx: 44 32
Softuart 1 Rx: 0
.Buffer idx: 45 32
Softuart 1 Rx: 0
.Buffer idx: 46 32
Softuart 1 Rx: 0
.Buffer idx: 47 32
Softuart 1 Rx: 0
.Buffer idx: 48 32
Softuart 1 Rx: 0
.Buffer idx: 49 32
Softuart 1 Rx: 0
.Buffer idx: 50 32
Softuart 1 Rx: 0
.Buffer idx: 51 32
Softuart 1 Rx: 0
.Buffer idx: 52 32
Softuart 1 Rx: 0
.Buffer idx: 53 32
Softuart 1 Rx: 0
.Buffer idx: 54 32
Softuart 1 Rx: 0
.Buffer idx: 55 32
Softuart 1 Rx: 0
.Buffer idx: 56 32
Softuart 1 Rx: 0
.Buffer idx: 57 32
Softuart 1 Rx: 0
.Buffer idx: 58 32
Softuart 1 Rx: 0
.Buffer idx: 59 32
Softuart 1 Rx: 0
.Buffer idx: 60 32
Softuart 1 Rx: 0
.Buffer idx: 61 32
Softuart 1 Rx: 0
.Buffer idx: 62 32
Softuart 1 Rx: 0
.Buffer idx: 63 32
Softuart 1 Rx: 0
.Buffer idx: 0 32
Softuart 1 Rx: 0
.Buffer idx: 1 32
Softuart 1 Rx: 0
.Buffer idx: 2 32
Softuart 1 Rx: 0
.Buffer idx: 3 32
Softuart 1 Rx: 0
.Buffer idx: 4 32
Softuart 1 Rx: 0
.Buffer idx: 5 32
Softuart 1 Rx: 0
.Buffer idx: 6 32
Softuart 1 Rx: 0
.Buffer idx: 7 32
Softuart 1 Rx: 0
.Buffer idx: 8 32
Softuart 1 Rx: 0
.Buffer idx: 9 32
Softuart 1 Rx: 0
.Buffer idx: 10 32
Softuart 1 Rx: 0
.Buffer idx: 11 32
Softuart 1 Rx: 0
.Buffer idx: 12 32
Softuart 1 Rx: 0
.Buffer idx: 13 32
Softuart 1 Rx: 0
.Buffer idx: 14 32
Softuart 1 Rx: 0
.Buffer idx: 15 32
Softuart 1 Rx: 0
.Buffer idx: 16 32
Softuart 1 Rx: 0
.Buffer idx: 17 32
Softuart 1 Rx: 0
.Buffer idx: 18 32
Softuart 1 Rx: 0
.Buffer idx: 19 32
Softuart 1 Rx: 0
.Buffer idx: 20 32
Softuart 1 Rx: 0
.Buffer idx: 21 32
Softuart 1 Rx: 0
scandone
.Buffer idx: 22 32
Softuart 1 Rx: 0
reconnect
f -240, .Buffer idx: 23 32
Softuart 1 Rx: 0
.Buffer idx: 24 32
Softuart 1 Rx: 0
.Buffer idx: 25 32
Softuart 1 Rx: 0
.Buffer idx: 26 32
Softuart 1 Rx: 0
.Buffer idx: 27 32
Softuart 1 Rx: 0
.Buffer idx: 28 32
Softuart 1 Rx: 0
.Buffer idx: 29 32
Softuart 1 Rx: 0
.Buffer idx: 30 32
Softuart 1 Rx: 0
.Buffer idx: 31 32
Softuart 1 Rx: 0
.Buffer empty
Softuart 1 Rx: 0
................................scandone
.reconnect
f 0, ..................................... 
```

Thanks in advance
mwolf

plieningerweb, 2015-07-21:
Hey @mwolfeu ,

thanks for your issue report. I have some more questions: 
-How did you wire up your esp?
-From where did you send data to your esp rx?

Hopefully with this information I will be able to reproduce the issue.
Thanks!

mwolfeu, 2015-07-21:
It is on a breakout board similar to this:
http://g04.a.alicdn.com/kf/HTB19bIBHVXXXXbdXFXXq6xXFXXXj/ESP8266-ESP-12-Serial-WIFI-Module-Industrial-Stable-Version-Test-Board-Full-IO-Leads.jpg

I used a loopback so the data came from pin 14 -> pin 12.

When I had pin 14 -> Rx on an external Uart everything looked great.  (received several '1's)

I hooked pin12 to TX on an USB to 3.3v Uart device with interesting results:
  The following happened when typed 0-9 by hand from my linux terminal
```
Softuart 1 Rx: 103
....Buffer empty
Softuart 1 Rx: 51
....Buffer empty
Softuart 1 Rx: 102
...Buffer empty
Softuart 1 Rx: 25
....Buffer empty
Softuart 1 Rx: 101
..scandone
.Buffer idx: 32 34
Softuart 1 Rx: 50
reconnect
f 0, .Buffer idx: 33 34
Softuart 1 Rx: 100
.Buffer idx: 34 35
Softuart 1 Rx: 12
.Buffer empty
Softuart 1 Rx: 99
...Buffer empty

```
Note what it thinks it receives.
'0' is ASCII 00110000   and it thinks it got decimal 103 which is 01100111

thanks,
mwolf

mwolfeu, 2015-07-21:
I just did some more testing with the external usb uart.   At 38400 soft uart seems solid.      
Testing further and making two soft uarts didn't help anything.

 I would think the goofy LEDs on my board might be screwing with the send/receive but if it makes any difference, I just tested them when on with my multimeter and got 3.26v.

plieningerweb, 2015-07-22:
@mwolfeu 

I am sorry to let you know that "looback" does not work. Sending data form one tx to another software rx was never able with this library.

I am even not sure, if other libraries, e.g. Arduino SoftwareSerial, can do this.

Sorry that this was not clear form the beginning.
Are there any other issues? If not, I will close this one

mwolfeu, 2015-07-22:
okay, I misunderstood your previous two software uart example. I thought it suggested loopback was possible. You are right about the Arduino. As I understand it, it polls the GPIO with interrupts off, so nothing else can happen during this time.
You can close this issue. Thanks again for the code!

closed 2015-07-22 by plieningerweb