plieningerweb/esp8266-software-uart #3: Error undefined reference to `PIN_PULLDWN_DIS' !!!

tinamore, 2015-08-14:
Hi
Thank you for great project
I complie esp8266-software-uart, But elipse show error:
----------
softuart.c:50: undefined reference to `PIN_PULLDWN_DIS'
....
softuart.c:129: undefined reference to `PIN_PULLDWN_DIS'
------------
I use SDK 1.2.0
Can you fix this erorr ?
thanks.

krzychb, 2015-09-19:
This error is due to changes introduced in esp_iot_sdk_v1.1.0_15_05_26.
For details please refer to http://bbs.espressif.com/viewtopic.php?f=5&t=481&p=1832
To fix it I commented out all occurrences of PIN_PULLDWN_DIS(...) in file softuart.c - there were three of them. 

Thanks plieningerweb - the library works great :-)

plieningerweb, 2015-09-27:
Just fixed the issue by removing all PIN_PULLDWN_DIS

closed 2015-09-27 by plieningerweb