plieningerweb/esp8266-software-uart #8: pins not working

scargill, 2016-02-16:
Got an issue maybe you can help. Using pins gpio4 and gpio5... we use gpio12 and gpio13 for something completely different.. When we init your software serial - just realised gpio12 and 13 stop working. Not using RS485. Does the serial code attempt to use any hardware lines for control etc.. and if so where so we can disable it???

Pete.

plieningerweb, 2016-03-01:
no, It actually should not use any hardware lines...
I am not sure what your issue is.
- is 12 and 13 no working anymore?
- did you not init them with the library? -> then it should not use them (there is nothing hard coded, etc..)

Is this still a issue? 
If yes, then please respond with more details or a solution.
Otherwise, we will close this soon.

plieningerweb, 2016-03-30:
Close issue because of missing details

closed 2016-03-30 by plieningerweb