plieningerweb/esp8266-software-uart #9: baudrate limit

dziadyga, 2016-08-21:
void Softuart_Init(Softuart *s, uint16_t baudrate)

"baudrate" parameter being uint16_t is limited to 65535.
Is this intentional? (not to use higher baud rates?)
After changing parameter type to "uint32_t" baud rate 115200 works (kinda).
Also - "bit_time" is calculated to 8 for baudrate 115200, but I think 9 would be closer.

plieningerweb, 2016-08-22:
This was not intentional and seems like a bug.  I was only skeptical if
these high rates will work.

So I think your findings are very valuable!

Could you maybe create a pull request? This would help me a lot!

plieningerweb, 2016-08-22:
@dziadyga , Thanks for the pull request! Just merged!

closed with #10

closed 2016-08-22 by plieningerweb