plieningerweb/esp8266-software-uart #14: issue in Softuart_Putchar

emichel11, 2017-05-05:
The for loop in Softuart_Putchar has an issue. It runs 9 times instead of 8.
it should read
for(i = 0; i < 8; i ++ )

plieningerweb, 2017-05-05:
Could you do a pull request?

emichel11, 2017-05-05:
It seems I do not have the permission to generate a pull request. My push fails.
Here the diff:
324c324
< 	for(i = 0; i <= 8; i ++ )
---
> 	for(i = 0; i < 8; i ++ )

emichel11, 2017-05-05:
It seems I do not have the permission. My push fails. Therefore here the diff:  
324c324
< 	for(i = 0; i <= 8; i ++ )
---
> 	for(i = 0; i < 8; i ++ )

plieningerweb, 2017-05-21:
Thanks for the issue!

Regarding the pull request: Usually you do first a fork on github, then push to your own repository, then do a pull request to this repository.

Thanks for reporting, let me know if the issue is solved now!

commit d01c4eae (2017-05-21): Fix: 8 Chars instead of 9 at a time

This closes #14
closed 2017-05-21 by plieningerweb